permissionlesstech/bitchat #23: migrating Bitchat Project to Flutter

aykutuludag, 2025-07-07:
Hi @jackjackbits . As you can see, your project has excited many people.

While the project is in its early stages, how about moving it to Flutter to support other platforms?

There are plugins available for this on the Flutter side (https://pub.dev/packages/bluetooth_low_energy)

I am currently trying to develop an app to test if it works or not, it will simply send text messages between devices via BLE. In the future, I will try to send small images, videos and audios. (https://github.com/aykutuludag/flutter_ble_mesh)

If we move to Flutter, we will be able to support Android, Desktop Platforms (macOS, Windows & Linux) and Embedded systems in addition to iOS. This will take the project to a whole new level.

Thanks to the codes you have already written, the iOS side is almost ready. With the help of the existing plugins, we can move the project to Flutter in a short time.

I wish you a nice week.

wacumov, 2025-07-07:
There's a great newer tool called [Skip](https://skip.tools/) that makes it relatively easy to add an Android client without rewriting the existing Swift codebase. 
Unlike Flutter, Skip produces two fully native apps - SwiftUI on iOS and Jetpack Compose on Android. 
You can see a clear comparison of different approaches [here](https://skip.tools/compare/).

jackjackbits, 2025-07-08:
thank you, but i'm going to stick with swift. i want it to be more native. the protocol will remain simple enough to create other native clients.

closed 2025-07-08 by jackjackbits