permissionlesstech/bitchat #3: Cool idea - Can't get it to connect between 2 of my devices

tsbernar, 2025-07-07:
I installed on 2 iOS devices with bluetooth on.  Opened the app on both and waited for 2 minutes.  Screenshot attached 
[image]

I like the idea

jackjackbits, 2025-07-08:
interesting. i need more info though.

tsbernar, 2025-07-08:
I built from source on a 3rd device (MacOS) and ended up connecting device 1 and device 3.  
Turns device 2 (iOS) had Bluetooth disabled in settings -> apps -> bitchat -> bluetooth (but it did have bluetooth on globally). 
User error here, but a possible QoL improvement would be showing a permission prompt to the user if Bluetooth isn't accessible.

tsbernar, 2025-07-08:
I will tinker with a PR for this, (but I am a swift noob)

tsbernar, 2025-07-08:
> I will tinker with a PR for this, (but I am a swift noob)

https://github.com/jackjackbits/bitchat/pull/37

closed 2025-10-07 by jackjackbits