permissionlesstech/bitchat #4: Determining location on Mesh Network

aykutuludag, 2025-07-07:
Hi @jackjackbits . First of all, I am quite pleased once I saw this project. I was working similar things a few years ago. I think this kind of technology will become very widespread in the next 5 years. I saw in the whitepaper that one of the aims of the project was to be able to communicate without an internet infrastructure.

```
WiFi Direct Transport
A planned enhancement will add WiFi Direct as an alternative transport layer:

100x bandwidth: 250+ Mbps vs BLE's 1-3 Mbps
Extended range: 100-200m vs BLE's 10-30m
Automatic handoff: Seamlessly switch between BLE and WiFi Direct
Hybrid mesh: Some nodes BLE-only, others WiFi-capable
Battery-aware: Only activate for large transfers or when charging
Alternative Transports
Future transport options being considered:

Ultrasonic Communication: 1-10 kbps through air, works when radio is jammed
LoRa (Long Range): 2-15km range for disaster scenarios
Transport bonding: Use multiple simultaneously for redundancy
```

This part is very cool. Because BLE has limitation. Even sending a simple image file with BLE is problematic. Hybrid mechacism is the best until BLE devices has wide range and bandwidths.

May I suggest can we add location feature to bitchat? Because in disaster scenarios, location is the most important information. Location can be determined by following methods:

**1) Device A has access to GPS**

If Device A has GPS, it can determine its location and send like WhatsApp or other messaging app doing.

**2) Device A has not access to GPS but one of the devices in the mesh has (Device B).**

In this scenario, Device A sends a message nearest devices name and their signal strength to network. When Device B got the message, it will calculate the distance of device A to itself and send it to the Device A. In this method we can calculate device A location by accuracy of (GPS + BLE distance calculation). For BLE distance calculation I found a formula like this:

```
 Distance Estimation Formula (Log-Distance Path Loss Model)

double rssi = -60;     // Measured RSSI in dBm
int txPower = -59;     // RSSI at 1 meter (calibrated)

double distance = Math.Pow(10, (txPower - rssi) / (10 * 2));

<html>
<body>
<!--StartFragment--><html><head></head><body>
RSSI (dBm) | Approximate Distance
-- | --
-40 | ~0.5 meters
-50 | ~1 meter
-60 | ~2–3 meters
-70 | ~5–7 meters
-80 | ~10–15 meters
-90 | >20 meters

</body></html><!--EndFragment-->
</body>
</html>
```

**3) None of the devices have not access to GPS.

In this scenario, we can only calculate device's position to each other in the mesh. We can't calculate GPS locations but calculate BLE locations which will be like.

Device A - You
Device B - 10 m away
Device C - 350m away

**4) Between mesh networks

Device A in MeshNetwork1 should able to send message to Device Z in MeshNetwork2. To able to do that, Device C (mediatory between 2 mesh Networks) should embed previous location inside itself's location. Calcations for this method can be found inside my code: https://github.com/aykutuludag/isaac-arduino/blob/master/isaac5/isaac5.ino

**5) WiFi Direct**

By wifi direct, we can determine location by wifi ssid. If we embed wifi locations inside app or it will be distributed by mesh network in the backstage, location can be determined (https://openwifimap.net)

Any communication system is layered with text, audio, video, and location. It is important to determine these communication protocols in the early stages of the system.

Have a nice day @jackjackbits and kudos for giving life to this project.

abdbali, 2025-07-07:
##  Feedback: Hardware compatibility suggestion for wider coverage and AM/FM bands

I am very happy to see a Turkish developer comment here! @aykutuludag 

Regarding AM-FM and mesh, referencing the compatibility of the following kits could greatly support users who want to reach wider coverage as well as experiment with FM-AM bands (which are not supported in current devices):

- **XIAO ESP32S3 & Wio-SX1262 Kit for Meshtastic**  
  [Seeed Studio Link](https://www.seeedstudio.com/Wio-SX1262-with-XIAO-ESP32S3-p-5982.html)

- **XIAO nRF52840 & Wio-SX1262 Kit for Meshtastic**  
  [Seeed Studio Link](https://www.seeedstudio.com/XIAO-nRF52840-Wio-SX1262-Kit-for-Meshtastic-p-6400.html)

These kits could enable experiments in achieving wider mesh network coverage and potentially integrating AM/FM capabilities in future revisions.

Thank you for your amazing work and looking forward to more innovative updates! 🚀

ameenbadri7, 2025-07-07:
Conversion Guide: Swift to Flutter
Here is a step-by-step guide to converting the "bitchat" project to Flutter.

**1. Set Up Your Flutter Environment**
If you haven't already, you'll need to set up your Flutter development environment. You can find detailed instructions on the official Flutter website.

**2. UI and Frontend**
The original app's UI is likely built with SwiftUI or UIKit. You will need to recreate this using Flutter's widget system.

Widget-Based UI: Flutter's UI is built with widgets. You will need to create a widget tree that mirrors the structure and appearance of the original app's UI. You can use a combination of StatelessWidget and StatefulWidget to build your UI components.

Navigation: Use a routing package like [go_router](https://pub.dev/packages/go_router) or Flutter's built-in Navigator to handle navigation between different screens of your app (e.g., from the main chat list to a specific chat room).

**3. Bluetooth Mesh Networking**
This is the most critical and challenging part of the conversion. You'll need to find Flutter packages that provide the necessary Bluetooth LE and mesh networking capabilities.

**Recommended Packages:**

- [flutter_reactive_ble](https://pub.dev/packages/flutter_reactive_ble): A popular package for handling Bluetooth Low Energy (BLE) communication.

- [flutter_blue_plus](https://pub.dev/packages/flutter_blue_plus): Another excellent option for BLE that is a fork and improvement of the now-unmaintained flutter_blue.

You will need to implement the following functionalities using one of these packages:

Device Scanning: Discover nearby devices that are also running your app.

Advertising: Make the user's device discoverable to others.

Connecting and Disconnecting: Establish and manage connections with other devices in the mesh network.

Data Transfer: Send and receive messages between connected devices.

**4. Cryptography**
The original project uses several cryptographic libraries for security. You will need to find equivalent packages in [Dart/Flutter](https://flutter.dev/) to implement the end-to-end encryption.

**Recommended Packages:**

- [pointycastle](https://pub.dev/packages/pointycastle): A comprehensive cryptography library for Dart that supports a wide range of algorithms.

- [cryptography](https://pub.dev/packages/cryptography): A modern and easy-to-use cryptography library for Flutter and Dart.

- [pinenacl](https://pub.dev/packages/pinenacl): A Dart implementation of the NaCl/libsodium cryptography library.

You will need to implement the following cryptographic features:

- Key Exchange: X25519

- Message Encryption: AES-256-GCM

- Password Hashing: Argon2id

- Digital Signatures: Ed25519

**5. Data Handling and Storage**
The "store and forward" functionality requires a local database to store messages when a recipient is offline.

**Recommended Packages:**

- [sqflite](https://pub.dev/packages/sqflite): A Flutter plugin for SQLite, a powerful local database.

- [sembast](https://pub.dev/packages/sembast): A NoSQL document-based database that is easy to use and works on all platforms.

You will need to design a database schema to store messages, chat rooms, and user information.

6. Platform Integration
Flutter's cross-platform nature is a huge advantage here. However, you'll need to handle platform-specific configurations, especially for Bluetooth permissions and background services.

- Permissions: Use the [permission_handler](https://pub.dev/packages/permission_handler) package to request the necessary Bluetooth and location permissions on both iOS and Android.

- Background Execution: To keep the mesh network running in the background, you may need to use a package like [flutter_background_service](https://pub.dev/packages/flutter_background_service).

**This conversion project is a significant undertaking, but by breaking it down into these steps, you can tackle it one piece at a time. Good luck! @aykutuludag**

jackjackbits, 2025-07-08:
yes, create a PR. not going to migrate to flutter however.

closed 2025-07-08 by jackjackbits

Peace-N, 2025-07-08:
This project should never be migrated to flutter but rather ffer a different flavour im suggesting a native androoid approach not Flutter::

https://github.com/Peace-N/bitchat-android