permissionlesstech/bitchat #7: Feedback: Font support suggestion

abdbali, 2025-07-07:
##  Feedback: Font support suggestion

As someone who has been interested in amateur radio, LoRa, and mesh for a while, this project fascinates me. My only visual suggestion will be about the font issue — I am eagerly waiting for the developments! It would be great if the software supports some fonts like our editors.

Here are some minimalist and highly readable font options that could be considered for future UI customization:

- **SF Pro Display** (Apple system font)  
  [Apple SF Pro](https://developer.apple.com/fonts/)

- **Roboto**  
  [Roboto on Google Fonts](https://fonts.google.com/specimen/Roboto)

- **Helvetica Neue**  
  [Helvetica Neue (Linotype)](https://www.linotype.com/1266/helvetica-neue-family.html)

- **Inter**  
  [Inter on GitHub](https://github.com/rsms/inter)

- **Space Grotesk**  
  [Space Grotesk on Google Fonts](https://fonts.google.com/specimen/Space+Grotesk)

---

Thanks for all your amazing work! Looking forward to seeing future updates.

accessor-io, 2025-07-07:
Sir, this is an iPhone Application.

> ## Feedback: Font support suggestion
>
> As someone who has been interested in amateur radio, LoRa, and mesh for a while, this project fascinates me. My only visual suggestion will be about the font [...]

Sir, this is an iPhone Application.

abdbali, 2025-07-07:
> Sir, this is an iPhone Application.
>
>
> Sir, this is an iPhone Application.

I know, I'm just suggesting fonts.

jackjackbits, 2025-07-08:
i think default monospaced font is good for now. can create a PR though.

closed 2025-07-08 by jackjackbits