permissionlesstech/bitchat #28: Quantum resistent cryptography

jessebs, 2025-07-08:
The current implementation uses Curve25519 for key exchange, which is not quantum-safe.

Consider adopting a quantum-resistant key exchange strategy. One potential approach is a hybrid protocol like [Signal’s PQXDH](https://signal.org/blog/pqxdh/), which combines classical and post-quantum cryptography.

accessor-io, 2025-07-11:
> The current implementation uses Curve25519 for key exchange, which is not quantum-safe.
>
> Consider adopting a quantum-resistant key exchange strategy. One potential approach is a hybrid protocol like [...]

I started to use Signals encryption method. 
Currently working on a new approach, under testing atm.

david415, 2025-07-12:
Perhaps you've heard of Wireguard VPN? Yes? They use Noise, specifically the IK handshake pattern which allows for zero round trip time encryption.

Here's some info on wireguard, it's legendary quality because of formalizing the cryptography in Tamarin:
https://www.wireguard.com/

Here's the Noise cryptographic protocol spec:
https://noiseprotocol.org/noise.html

Note that I don't recommend Noise, instead I recommend PQ Noise, I hope that's clear. However, later on when Trevor Perrin upgrades the Noise protocol to use KEMs then that might be something to consider.

Also note that you do not need to use an existing Noise library in order to use Noise; look at the wireguard example, Jason wrote his own implementation of the Noise handshake pattern that he wanted his protocol to use.

closed 2026-07-10 by jackjackbits

AnnoyedUnicorn, 2026-11-08:
Why was this issue closed as "completed" when it is not in fact completed?
The current Noise implementation still uses quantum-vulnerable EC25519 algorithm.

Also, the `BRING_THE_NOISE.md` file states "Kyber integration plans". Planning something is not the same as implementing it, so I think this issue should stay open until the integration has been made.
(Or another one regarding a full PQC compliance.)

In the meantime, I also think users should be made aware of the harvest-now, decrypt-later attack. (As does OpenSSH when using pre-quantum algorithms for instance.)