permissionlesstech/bitchat #42: Add Pre-Communication Identity Verification (QR Code / Public Key Fingerprint Binding)

castroi, 2025-07-08:
### **Problem** 
Imagine you're at a protest, conference, or disaster zone using **bitchat**. You see someone named "Bob" in the chat and start talking to them. But what if there's **another person** in the same room also using the name "Bob"? Or worse, what if someone is **pretending** to be your friend Bob from yesterday?

Right now, the app doesn’t have a way to **double-check identities** before sending your first message. Nicknames are easy to fake, and peer IDs reset every time the app restarts. So if you haven’t marked Bob as a favorite before, **you can’t be 100% sure** who you’re talking to.
****
### **Proposed Solution**
Add a **manual identity verification flow** for first-time communication. This would allow users to:
1. **Scan a QR code** or **exchange public key fingerprints** in person to cryptographically bind identities.
2. Use this binding to **mark verified peers** and display warnings for unverified ones.

### **Implementation Ideas**
- **QR Code Verification**:
  - When Alice and Bob meet, one generates a QR code containing:
    - Public key fingerprint (e.g., `SHA-256(publicKey)`).
    - Nickname.
    - Timestamp (to prevent replay attacks).
  - Scanning the QR code adds the peer to Alice’s "verified favorites" list.
- **Public Key Display**:
  - Show public key fingerprints in the peer profile (e.g., `0x1a2b3c4d...`).
  - Allow manual comparison (e.g., "My key ends with `3c4d`, does yours?").
- **Verification Flow**:
  - Before sending a message to a new peer, prompt Alice to:
    - "Verify this peer’s identity to prevent impersonation."
    - Options: **Scan QR**, **Compare Fingerprint**, or **Send Anyway (Unverified)**.
  - Verified peers get a "✓ Verified" badge in the UI.

****

ND255, 2025-07-08:
полная хуйня

castroi, 2025-07-08:
@ND255, Thanks for the feedback!

castroi, 2025-07-31:
@zabojeb where can I find this fingerprint screen in the iOS app?