permissionlesstech/bitchat #56: Is it possible to add lora devices?

xanderolo, 2025-07-08:
Is it possible to make it hybrid with lora? Meshtastic/meshcore devices are able to extend communication.

thegoatmilkguy, 2025-07-08:
It would be awesome if this could hook existing meshtastic infrastructure.  That way I could keep using my existing meshtastic gadgets connected to my phone via Bluetooth but also talk to other devices directly via Bluetooth.  Perhaps devices like this could act as "gateways" to meshtastic devices as long as the battery was high.

Pmheart6, 2025-07-12:
> Lora or Meshtastic functionality would make this project more useful and interesting.
> 
> A quick internet search shows that many people have made Bluetooth mesh chat apps over the years, none of which have "taken off" or been useful in my opinion. One such example: https://github.com/chrisballinger/BLEMeshChat
> 
> But I'm sure you know that, so maybe there's something else in store not yet mentioned…

Name recognition,  money, experience.  And a name that misses the grassroots aspect of tech so I highly doubt he'll commercialize it (like WhatsApp creator switching to signal)

I mean the talk has only been on a short while, yet there's a big buzz