permissionlesstech/bitchat #57: Android version?

hieblmi, 2025-07-08:
Is someone already porting the protocol to Android?

mbmaciel, 2025-07-08:
Android version please!

krystian3w, 2025-07-08:
Sad: https://github.com/jackjackbits/bitchat/issues/9#issuecomment-3046969503

alibektas, 2025-07-08:
check this one out https://github.com/callebtc/bitchat-android

vhbui02, 2025-07-09:
I think we should stay inside Swift ecosystem. It's very likely that @jackjackbits and his team will continue using Swift. They have deeper understanding on their project and we should follow their lead.

Lucky for us, Swift official support for Android is coming near ([Android Workgroup announcement 2 weeks ago](https://forums.swift.org/t/announcing-the-android-workgroup/80666)). Right now, our best bet is to wait, or learn Swift, join the workgroup and proactively contribute to the [roadmap](https://github.com/orgs/swiftlang/projects/17)

b-Istiak-s, 2025-07-10:
> Yes agreed, [@jackjackbits](https://github.com/jackjackbits) can you make a starting repo for this?

I think we should start the Android application once Bitchat is stable enough. @owenselles 
I  was thinking of building a project in Kotlin; however, the protocol, encryption and other segments will get updated very frequently,  so if we start the android app now, it will have to be continuously updated to match the requirements to communicate with Apple devices.

jackjackbits, 2025-07-10:
we now have an android repo that works. https://github.com/permissionlesstech/bitchat-android

closed 2025-07-10 by jackjackbits