permissionlesstech/bitchat #80: Upgrade to post quantum cryptography

david415, 2025-07-08:
Everything should be post quantum cryptography nowadays. And by that we really mean hybrid constructions.

jessebs, 2025-07-09:
See #28

david415, 2025-07-12:
Firstly, Wrong. Nothing Signal Messenger has done is a good example of post quantum cryptographic protocols. That thing you are pointing to, post quantum triple diffiehellman is a hybrid post quantum key exchange only. What Signal Messenger should have done was to upgrade to a KEM double ratchet which uses a hybrid post quantum key exchange for every message sent instead of just the initial key exchange.

Secondly, Signal is going for "post compromise security". If that is not the goal of your protocol that it would be inappropriate for your protocol to use a double ratchet (be it a NIKE double ratchet or a hybrid post quantum KEM double ratchet).

Probably you should hire someone like me to help you make these decisions because if you have people just pointing "oh look, Signal uses that stuff so maybe we should use that stuff too" but without an actual thought about the threat model.