permissionlesstech/bitchat #137: Standalone server based on nRF52840

ddamianus, 2025-07-10:
It would be great to have standalone server based on nRF52840 for app likde this.

It coud be on strategic position with solar or battery to coverrage good are. :)

loftafi, 2025-07-11:
I am exploring a port of the algorithm that could work on ESP32 and related devices. It would allow creating simple relays.  It's not ready yet though.

ddamianus, 2025-07-18:
That would be really great! Esp32 does have quite a big power consuption, but I'm not able to estimate the server requirements. However nRF would be ideal in terms of power consumption and running on battery or solar.

Is there i git code for testing?

Pucciano, 2025-07-23:
It would be nice to have some kind of stand alone repeater for BitChat to extend the coverage of the network.