permissionlesstech/bitchat #153: ‼️ App on Apple App Store misuses Icon and Name

benjamin-hg, 2025-07-10:
https://apps.apple.com/de/app/bitchat-social-app/id6741421002

[image]

kengovaert, 2025-07-10:
[https://apps.apple.com/be/app/bitchat-social-app/id6741421002](url) 

[image]

Same here in the App Store in Belgium

Reelix, 2025-08-06:
I suspect this one on the [Google Play store](https://play.google.com/store/apps/details?id=com.app.bitchat) as well.

[image]