permissionlesstech/bitchat #156: Here come the impersonators

thevalo, 2025-07-10:
I mean, it was a matter of time. There's this site, "bit-chats dot com" which is impersonating the bitchat. If you want to download it, it redirects to bitchatapk dot com, and wants you to download an apk. Here's the VT: https://www.virustotal.com/gui/file/7cd0cf81bc14498f71a540c534bc7785ede50f559a006079b52d357abab82fb2/details

Maybe it's a good idea to add a disclaimer in README that those pages do not represent the original bitchat?

I haven't been able yet to analyze it for its behavior and the objective. But it doesn't seem good.

SubatomicPlanets, 2025-07-10:
Weird thing to make fake apps for... Pretty much everyone using this knows not to install APKs from random websites. #153 is also a problem. Yeah a disclaimer would maybe be nice.

netpawn, 2025-07-10:
It was only a matter of time. Report domain, reclassify as infection point, vote down on community VT and yeah I guess add a disclaimer? 

As someone rightfully pointed out, people using this generally know not to install random APK.

ronibhakta1, 2025-07-10:
> I mean, it was a matter of time. There's this site, "bit-chats dot com" which is impersonating the bitchat. If you want to download it, it redirects to bitchatapk dot com, and wants you to download [...]

Exactly i was searching on Google and found the same site, but I was a bit unsure about how it is. So i checked for the GitHub repo. @jackjackbits please look into this matter

carbon-starlight, 2025-07-10:
I've got this website as the first result in Google, this GitHub page was only second. As due to how search ranking is structured first results in Google are usually 99% trustworthy, I can see many people falling for it. There is a chance I would, if not the unusual domain.

[image]

iDoNotConsentToBeingTracked, 2025-07-11:
@thevalo wait until you lookup the server IP address ...

josenicomaia, 2025-07-16:
Yeah, I was just checking here to match the site ownership before instaling on Android haha

A disclamer is going to be awesome

carbon-starlight, 2025-07-24:
Now the first result is Android app page. We don't have an Android app, right?

[image]

carbon-starlight, 2025-07-24:
Bing still has scam websites first. Can't we do anything about it? There surely has to be a mechanism for dealing with these?

[image]