permissionlesstech/bitchat #172: ❌ Build Failed on Mac: iOS storyboards do not support target device type "mac".

0xharkirat, 2025-07-11:
I have followed all the steps in setup:  
when I run on iOS, it works fine, but when I try to build for macOS, I get this error:

> ❌ `iOS storyboards do not support target device type "mac".`

[image]

Any suggestion is welcome.

Thanks!  
**PS:** Awesome app btw.

0xharkirat, 2025-07-15:
Any updates. 
Thanks
-h

kswzzl, 2025-07-25:
I'm not sure why this happens either, so I run it with `Option 4: just` from the README:

- `brew install just`
- `just run`

If you get the error `xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance`, run `sudo xcode-select -s /Applications/Xcode.app/Contents/Developer` then try again.

aroumie1997, 2025-07-29:
appstore build bug with keyboard

[image]

ignaciocervino, 2025-07-29:
> appstore build bug with keyboard
> 
> [image]

I submitted a [PR](https://github.com/permissionlesstech/bitchat/pull/336) that should fix this