permissionlesstech/bitchat #182: Will this App ever go to iOS App store?

koolninad, 2025-07-11:
Hey Bitchat community,

We're working on getting Bitchat on the App Store, but our initial submission for version 1.0 received feedback on Guideline 1.2 - User Generated Content. This is a critical hurdle we need community input on, especially considering Bitchat's decentralized nature.

The main issues are:

"Chatroulette-style" connections: Apple requires showing user info and giving accept/decline options before random connections. This directly challenges our current design.

Missing standard UGC features: We need to implement a formal EULA, content filtering, reporting, blocking, and commit to 24-hour response times for reports.

Our key questions to you are:

How can we integrate these App Store requirements (pre-chat user info, content moderation, reporting, blocking, EULA) while maintaining Bitchat's core principles of privacy and decentralization?

Are there any decentralized chat apps that have successfully met these specific App Store guidelines? What approaches did they use?

Your ideas and technical solutions are crucial for Bitchat's iOS launch.

prusnak, 2025-07-11:
You should use https://freedomstore.io (Alternative iOS AppStore focused on Freedom Tech) to distribute the App to the EU users seamlessly.

All you need is to send a PR to https://github.com/freedomstore/freedomstore/blob/main/altstore-source.json

jackjackbits, 2025-07-11:
yes, we will get it there. i want to make the changes outlined in the announcement post. are you trying to submit a forked version?

iDoNotConsentToBeingTracked, 2025-07-11:
You the goat, king 🐐👑 @jackjackbits

koolninad, 2025-07-12:
> yes, we will get it there. i want to make the changes outlined in the announcement post. are you trying to submit a forked version?

Hi Jack,
Thanks for your response.

Yes, I tried submitting a forked version on both iOS and Android. The iOS version was rejected, but surprisingly, the Android version got approved—even with everything kept the same for initial testing. To my surprise, it got over 25,000 downloads within the first 24 hours.

I'm now in the process of changing the name and branding to make it clearly distinct and avoid confusion. I really appreciate the work you've done with the original project and I’m excited to see where it goes, especially with the upcoming iOS launch.

Just wanted to be transparent about my testing and share the unexpected traction the fork got. Happy to align in any way that keeps things respectful to the original effort.

JMcrafter26, 2025-07-12:
Hi @koolninad and Bitchat team,
thanks for the update on App Store requirements!

I don’t currently have a TestFlight slot or access to a Mac, and I’d love to beta test the iOS version if possible. Would you, or anyone in the community, be able to provide a compiled IPA for sideloading - or share guidance on how to build it from source on non‑macOS systems (maybe even through GitHub Actions)?

I'd really appreciate any help or direction. 🙏

koolninad, 2025-07-13:
> Hi [@koolninad](https://github.com/koolninad) and Bitchat team, thanks for the update on App Store requirements!
>
> I don’t currently have a TestFlight slot or access to a Mac, and I’d love to beta [...]

No man, you cannot compile iOS version without a Mac. Try alternative code on https://freedomstore.io/ or Hetzner cloud offers Mac Mini in the cloud, but it is very slow you can use it for compiling purposes

xanderolo, 2025-07-13:
> No man, you cannot compile iOS version without a Mac. Try alternative code on https://freedomstore.io/ or Hetzner cloud offers Mac Mini in the cloud, but it is very slow you can use it for compiling purposes

Is it possible to compile it wihout paid apple developer account?

KingOfMac, 2025-07-14:
> 
> > No man, you cannot compile iOS version without a Mac. Try alternative code on https://freedomstore.io/ or Hetzner cloud offers Mac Mini in the cloud, but it is very slow you can use it for compiling purposes
> 
> Is it possible to compile it wihout paid apple developer account? 
> 
> 

With a free apple developer id you can get a 7 day provisioning profile

KingOfMac, 2025-07-14:
Your certificate is added before compiling

KingOfMac, 2025-07-14:
Exactly.

dogederp, 2025-08-04:
Repository owners are neglecting basic issue management