permissionlesstech/bitchat #183: Multidevice identity key export/import via QR or seed, peerID / hash name across devices.

mstiv, 2025-07-11:
**Problem**

Right now BitChat locks my identity to a single device. If I lose or switch phones I have to make a new key and lose my own chats and saved contacts, and everyone else ends up seeing 2–3 versions of me in their list. It gets messy fast.

**What I'd like to build (if the community is ok with it)**

1. Backup / Export identity: 
- add a Settings button to show either
- a 24-word seed or
- a single encrypted QR code (< 600 bytes).

2. Import / Restore - on a fresh install, user scans the QR or enters the seed and gets the same key.

3. Use the same identity on multiple devices:
- Allow importing the same seed on a tablet or laptop so all devices send as one account.
- Each device could sign with a derived sub-key so revoking a lost phone is possible.

4. Optional passphrase to encrypt the backup for extra safety.

**Why it matters**

1. Stops “ghost” duplicate entries that look like fake accounts.
2. Lets people run BitChat on multiple gadgets just like Signal or WhatsApp.

If others think this is useful, I’m happy to start working on it!

mstiv, 2025-07-11:
This product really hooked me. I can't think about anything else. I'm testing BitChat on different platforms. I believe it's the next step for secure communication and a stronger crypto community.

vizzv, 2025-07-11:
Just chiming in — I think this is a really valuable idea.
Backup and multi-device support are badly needed, and I love the approach of using a seed phrase or encrypted QR code, similar to crypto wallets.

However, one thing I’m curious (and slightly concerned) about:
If someone else somehow gets access to the seed phrase or QR backup, wouldn’t they be able to restore the account on their device too? That could let them impersonate the user or even spy on chats .

In other words, multi-device support also increases the risk of key leakage leading to identity theft.

Possible ideas to mitigate this:

- Notify original devices if a new device joins
- Ability to revoke devices quickly

Curious to hear what others think about this balance between usability and security!