permissionlesstech/bitchat #220: bitchat apk

Rumam786, 2025-07-12:
How to install bitchat app for android ?

iDoNotConsentToBeingTracked, 2025-07-12:
Open ADB
Then sit down, cross your legs and pray to Jesus.

JMcrafter26, 2025-07-12:
> How to install bitchat app for android ?

Read this: https://github.com/permissionlesstech/bitchat-android#install-bitchat

travnewmatic, 2025-07-15:
Also, is this a legitimate source for the android APK? https://bitchatapk.com/

rsaccon, 2025-07-16:
Oh, that one is 12 MB big, according to the website. That is about four times the file size of the legitimate one, with other words: it might contain 9 MB of malware!

travnewmatic, 2025-07-16:
Yeeep that was my concern :(

Got the official one from your other repo (now)

I'm an idiot and should have read more but it might beneficial if download links were front and center on any/all of the GitHub readmes

travnewmatic, 2025-07-16:
There are *a lot* of people domain squatting and pushing their own flavors of bitchat APKs.  It's kind of terrifying.

Is there an official bitchat homepage or is that this GitHub?  (It might be worth mentioning that this is the only legitimate APK source and to avoid any others)