permissionlesstech/bitchat #222: [FEATURE] 🌍 Localization Support Request

wiseweb-works, 2025-07-13:
**What's needed and why?**

As someone who cares deeply about open source software and user accessibility, I’d like to help you add localization (foreign language support) to your project. I’ve previously contributed to similar projects on GitHub on a voluntary basis.

Since **BitChat** is a mobile application (Android/iOS), offering multi-language support can significantly improve the experience for non-English-speaking users. This would make the app more inclusive and easier to use globally.

**Implementation ideas (optional)**

My only request is that you structure your user-facing strings in a way that supports external translation files — similar to how other projects handle localization. I can start by translating the app into my native language, and later, other contributors can easily add more translations without modifying the codebase.

The key idea is:

* Extract all user interface text into separate language files (e.g. JSON, XML, or strings files).
* Dynamically load strings based on the device or app’s selected language.

This approach keeps the source code clean while enabling seamless language switching.

**Helping localization**

To make the translation process easier and more collaborative, I suggest registering your project on [Weblate](https://hosted.weblate.org/) — a free, open-source localization platform. It allows volunteer translators to contribute easily and helps manage translations more efficiently. There’s no cost involved, and it aligns well with open source principles.

**Reference projects using this model:**

* [RSStT Bot](https://github.com/Rongronggg9/RSS-to-Telegram-Bot/blob/dev/docs/translation-guide.md)
* [Monal (XMPP app)](https://github.com/monal-im/Monal)
* [Archinstall](https://github.com/archlinux/archinstall/releases/tag/v2.5.0)
* [Mastodon-T-Crossposter](https://github.com/renatolond/mastodon-twitter-poster)
* [Cuppa by Nathanatos](https://www.nathanatos.com/)
* [Bunkerweb](https://github.com/bunkerity/bunkerweb)
* [SimpleX-Chat](https://github.com/simplex-chat/simplex-chat)

I’d be happy to contribute translations. Thank you for considering!

hadymaggot, 2025-07-13:
I ma ready for Indonesian language

ddamianus, 2025-07-18:
Ready for Czech

closed 2025-12-03 by wiseweb-works