permissionlesstech/bitchat #225: Replace ECC (quantum vulnerable cryptography) with Kyber, Dilithium, AES-256 for quantum secure messaging

successor1, 2025-07-13:
## Quantum resistance

For long-term protection and make bitchat last for decades - im suggesting we are introducing quantum resistance to it e.g. no capture now, decrypt later issues etc. 

## Current way

Private Messages: X25519 key exchange + AES-256-GCM encryption
Channel Messages: Argon2id password derivation + AES-256-GCM
Digital Signatures: Ed25519 for message authenticity
Forward Secrecy: New key pairs generated each session

## My suggestions

Private Messages: **Kyber** key exchange + AES-256-GCM encryption
Channel Messages: **Argon2id password derivation (quantum vulnerable, needs replacement)** + AES-256-GCM
Digital Signatures: **Dilithium** for message authenticity
Forward Secrecy: New key pairs generated each session **(no changes)**

successor1, 2025-07-20:
> A hibrid sistem is better, PQ-safe cyphers are much new to solo use, hibrid sistems are more Safe. 
> 
> ECC + KYBER & DILITHIUM.

Yes - that is fine with me. Same results 👍🫡

BigPanda97, 2025-07-21:
> Channel Messages: Argon2id password derivation (quantum vulnerable, needs replacement)

Argon2 is symmetrical, so since when is it quantum vulnerable?!

> AES-256-GCM encryption

On AES, up to 13 out of 14 rounds can be attacked by some scenarios. For PQ security, combining AES and ChaCha20 would be good.
Speed as well as energy consumption does not make a real difference / is definitely NOT noticeable here.