permissionlesstech/bitchat #276: Add support to offline Monero (XMR) payments

katipiry, 2025-07-20:
[Monero](https://getmonero.org) is a cypherpunk-based cryptocurrency based in Cryptonote protocol, Monero is like Bitcoin (PoW consensus, P2P and decentralized), but Monero have privacy to amount and sender (with RingCT, FCMP++ in future), receiver adress protection (Stheath adress) and sender IP protection (dandelion++).

**Why?**
* Private transactions; 
* Support to a fungible decentralized currency; 
* More cypherpunk users will use BitChat; 
* More decentralization of Monero network (no needs Wi-Fi to make a transaction); 
* Support connection with user node(s) without needs a remote node (protection against spy-nodes).

The problems: 
* Transaction size (after FCMP++ the size of transactions will increase to ~4Kb); 
* More complex to implement without Wi-Fi;
* 10-block limit (the transactions have ~20 minutes to be confirmed).

successor1, 2025-07-20:
Good idea. Might as well implement Tornado Cash for transaction specific anonymity for the masses.

katipiry, 2025-07-20:
> Good idea. Might as well implement Tornado Cash for transaction specific anonymity for the masses.

Or both, but I believe Monero is better because:

* All transactions are anonymous (no different transactions);
* FCMP++ will improve Monero on a divine level (giving some quantum protection, of course, it's not ideal, but Forward Secrecy is already a very good improvement in quantum terms).
* Monero is a cryptocurrency in its own right, it doesn't depend on an external tool, Monero is a more audited school and I would say more trustworthy (or rather, it needs less trust than TornadoCash).

But there's nothing stopping you from implementing both (just the complexity, of course).

successor1, 2025-07-20:
> Or both, but I believe Monero is better because:
>
> * All transactions are anonymous (no different transactions);
> * FCMP++ will improve Monero on a divine level (giving some quantum protection, of [...]

I like the unlinkability feature of Tornado Cash, the only issue is that anything coming out of it gets flagged. A simple act of resistance would be to make more people use it, similar to Tor. The more people use it, the less likely those flags will have any meaning to it at all 🧠

successor1, 2025-07-20:
> > > > Good idea. Might as well implement Tornado Cash for transaction specific anonymity for the masses.
> > > 
> > > 
> > > Or both, but I believe Monero is better because:
> > > 
> > > * All transactions are anonymous (no different transactions);
> > > * FCMP++ will improve Monero on a divine level (giving some quantum protection, of course, it's not ideal, but Forward Secrecy is already a very good improvement in quantum terms).
> > > * Monero is a cryptocurrency in its own right, it doesn't depend on an external tool, Monero is a more audited school and I would say more trustworthy (or rather, it needs less trust than TornadoCash).
> > > 
> > > But there's nothing stopping you from implementing both (just the complexity, of course).
> > 
> > I like the unlinkability feature of Tornado Cash, the only issue is that anything coming out of it gets flagged. A simple act of resistance would be to make more people use it, similar to Tor. The more people use it, the less likely those flags will have any meaning to it at all 🧠
> 
> I understand your point and I agree, but there are some big differences:
> * Monero is already Decentralized by default;
> * All transactions are already fungible (i.e. there are no "suspicious" transactions, they are all the same); 
> * Monero has a proven track record of thwarting government agencies and ChainAnalysis;
> * Monero after FCMP++ will have a gigantic anonymity pool (more than 100 million), plus the community is driven by long-time cypherpunks and people at risk of their lives who need their identities protected.

I agree with your points 🧠, both may be used, yes. The benefit of Tornado Cash is EVM compatability and outside the monero blockchain. Monero is great itself, no problems with that 👍

KingOfMac, 2025-07-21:
Great idea. I think @jackjackbits said he wanted bitcoin intergration also

katipiry, 2025-07-21:
> Great idea. I think [@jackjackbits](https://github.com/jackjackbits) said he wanted bitcoin intergration also

Yes, from what I understand he wants to integrate Bitcoin into BitChat, and I see this as a great possibility of integrating Monero (XMR) into BitChat, that's going to be incredible.

I really want to see Monero (XMR) implemented on BitChat, and the best part is that I know I'm not the only one who wants it.

shortwavesurfer2009, 2025-07-21:
I would love to see this and I'll bet the Monero community would be willing to donate to a bounty to get this funded on bounties.monero.social