permissionlesstech/bitchat #283: Remove compression for privacy

BigPanda97, 2025-07-21:
I'm not good at explaining the background behind this, so I let ChatGPT explain to you , why you should instantly remove compression (LZ4 and other algorithms) from the protocol:

<html><body>
<p data-start="0" data-end="710"><strong data-start="0" data-end="13">In short:</strong> When you compress data <em data-start="37" data-end="45">before</em> you encrypt it, the size of the ciphertext is no longer fixed—it shrinks in proportion to how much redundancy the compressor could eliminate. Because the attacker can <em data-start="213" data-end="265">see that length even though the bits are encrypted</em>, they gain a “compression‑ratio side‑channel.” If they can repeatedly influence part of the plaintext (for example by injecting guesses into a request) and observe how the length changes, they can gradually learn the secret portions byte‑by‑byte. Real‑world exploits such as <strong data-start="541" data-end="550">CRIME</strong>, <strong data-start="552" data-end="562">BREACH</strong>, and <strong data-start="568" data-end="579">VORACLE</strong> have shown that this leaks cookies, CSRF tokens, and even VPN traffic despite strong ciphers.<span class="" data-state="closed"><span class="ms-1 inline-flex max-w-full items-center relative top-[-0.094rem] animate-[show_150ms_ease-in]"><a href="https://en.wikipedia.org/wiki/CRIME?utm_source=chatgpt.com" target="_blank" rel="noopener" alt="https://en.wikipedia.org/wiki/CRIME?utm_source=chatgpt.com" class="flex h-4.5 overflow-hidden rounded-xl px-2 text-[9px] font-medium text-token-text-secondary! bg-[#F4F4F4]! dark:bg-[#303030]! transition-colors duration-150 ease-in-out"><span class="relative start-0 bottom-0 flex h-full w-full items-center"><span class="flex h-4 w-full items-center justify-between overflow-hidden"><span class="max-w-full grow truncate overflow-hidden text-center">Wikipedia</span></span></span></a></span></span><span class="" data-state="closed"><span class="ms-1 inline-flex max-w-full items-center relative top-[-0.094rem] animate-[show_150ms_ease-in]"><a href="https://www.breachattack.com/?utm_source=chatgpt.com" target="_blank" rel="noopener" alt="https://www.breachattack.com/?utm_source=chatgpt.com" class="flex h-4.5 overflow-hidden rounded-xl px-2 text-[9px] font-medium text-token-text-secondary! bg-[#F4F4F4]! dark:bg-[#303030]! transition-colors duration-150 ease-in-out"><span class="relative start-0 bottom-0 flex h-full w-full items-center"><span class="flex h-4 w-full items-center justify-between overflow-hidden"><span class="max-w-full grow truncate overflow-hidden text-center">breachattack.com</span></span></span></a></span></span><span class="" data-state="closed"><span class="ms-1 inline-flex max-w-full items-center relative top-[-0.094rem] animate-[show_150ms_ease-in]"><a href="https://openvpn.net/security-advisory/the-voracle-attack-vulnerability/?utm_source=chatgpt.com" target="_blank" rel="noopener" alt="https://openvpn.net/security-advisory/the-voracle-attack-vulnerability/?utm_source=chatgpt.com" class="flex h-4.5 overflow-hidden rounded-xl px-2 text-[9px] font-medium text-token-text-secondary! bg-[#F4F4F4]! dark:bg-[#303030]! transition-colors duration-150 ease-in-out"><span class="relative start-0 bottom-0 flex h-full w-full items-center"><span class="flex h-4 w-full items-center justify-between overflow-hidden"><span class="max-w-full grow truncate overflow-hidden text-center">OpenVPN</span></span></span></a></span></span></p>
<hr data-start="712" data-end="715">
<h2 data-start="717" data-end="752">1.  Where the leakage comes from</h2>
<h3 data-start="754" data-end="806">1.1 Variable‑length output survives encryption</h3>
<ul data-start="807" data-end="1543">
<li data-start="807" data-end="983">
<p data-start="809" data-end="983"><strong data-start="809" data-end="844">Compression removes redundancy.</strong> Repeated strings (e.g., “sessionid=ABC…” appearing twice) encode to fewer bits than unique strings.<span class="" data-state="closed"><span class="ms-1 inline-flex max-w-full items-center relative top-[-0.094rem] animate-[show_150ms_ease-in]"><a href="https://crypto.stackexchange.com/questions/38047/compression-ratio-side-channel?utm_source=chatgpt.com" target="_blank" rel="noopener" alt="https://crypto.stackexchange.com/questions/38047/compression-ratio-side-channel?utm_source=chatgpt.com" class="flex h-4.5 overflow-hidden rounded-xl px-2 text-[9px] font-medium text-token-text-secondary! bg-[#F4F4F4]! dark:bg-[#303030]! transition-colors duration-150 ease-in-out"><span class="relative start-0 bottom-0 flex h-full w-full items-center"><span class="flex h-4 w-full items-center justify-between overflow-hidden"><span class="max-w-full grow truncate overflow-hidden text-center">Cryptography Stack Exchange</span></span></span></a></span></span></p>
</li>
<li data-start="984" data-end="1215">
<p data-start="986" data-end="1215"><strong data-start="986" data-end="1030">Encryption hides content but not length.</strong> Stream‑ and AEAD‑ciphers keep the byte‑count unchanged, so the ciphertext length equals the compressed length plus a small, predictable overhead.<span class="" data-state="closed"><span class="ms-1 inline-flex max-w-full items-center relative top-[-0.094rem] animate-[show_150ms_ease-in]"><a href="https://www.iacr.org/cryptodb/archive/2002/FSE/3091/3091.pdf?utm_source=chatgpt.com" target="_blank" rel="noopener" alt="https://www.iacr.org/cryptodb/archive/2002/FSE/3091/3091.pdf?utm_source=chatgpt.com" class="flex h-4.5 overflow-hidden rounded-xl px-2 text-[9px] font-medium text-token-text-secondary! bg-[#F4F4F4]! dark:bg-[#303030]! transition-colors duration-150 ease-in-out"><span class="relative start-0 bottom-0 flex h-full w-full items-center"><span class="flex h-4 w-full items-center justify-between overflow-hidden"><span class="max-w-full grow truncate overflow-hidden text-center">iacr.org</span></span></span></a></span></span></p>
</li>
<li data-start="1216" data-end="1543">
<p data-start="1218" data-end="1543"><strong data-start="1218" data-end="1276">Length therefore correlates with plaintext similarity.</strong> If an attacker’s guess about a secret makes the surrounding text more redundant (e.g., their guess matches a cookie fragment already in the message), the compressed output becomes shorter, revealing that the guess was “closer.”<span class="" data-state="closed"><span class="ms-1 inline-flex max-w-full items-center relative top-[-0.094rem] animate-[show_150ms_ease-in]"><a href="https://crypto.stackexchange.com/questions/38047/compression-ratio-side-channel?utm_source=chatgpt.com" target="_blank" rel="noopener" alt="https://crypto.stackexchange.com/questions/38047/compression-ratio-side-channel?utm_source=chatgpt.com" class="flex h-4.5 overflow-hidden rounded-xl px-2 text-[9px] font-medium text-token-text-secondary! bg-[#F4F4F4]! dark:bg-[#303030]! transition-colors duration-150 ease-in-out"><span class="relative start-0 bottom-0 flex h-full w-full items-center"><span class="flex h-4 w-full items-center justify-between overflow-hidden"><span class="max-w-full grow truncate overflow-hidden text-center">Cryptography Stack Exchange</span></span></span></a></span></span></p>
</li>
</ul>
<h3 data-start="1545" data-end="1592">1.2 The “compression‑oracle” attack model</h3>
<ol data-start="1593" data-end="1928">
<li data-start="1593" data-end="1682">
<p data-start="1596" data-end="1682">Attacker injects chosen text next to an unknown secret (e.g., in a query parameter).</p>
</li>
<li data-start="1683" data-end="1761">
<p data-start="1686" data-end="1761">Server compresses <strong data-start="1704" data-end="1730">secret + attacker text</strong>, then encrypts and sends it.</p>
</li>
<li data-start="1762" data-end="1835">
<p data-start="1765" data-end="1835">Attacker measures the encrypted packet length (a free side‑channel).</p>
</li>
<li data-start="1836" data-end="1928">
<p data-start="1839" data-end="1928">By iterating guesses, they reconstruct the secret.<span class="" data-state="closed"><span class="ms-1 inline-flex max-w-full items-center relative top-[-0.094rem] animate-[show_150ms_ease-in]"><a href="https://crypto.stackexchange.com/questions/15138/how-does-compression-before-encryption-leak-info-about-the-input?utm_source=chatgpt.com" target="_blank" rel="noopener" alt="https://crypto.stackexchange.com/questions/15138/how-does-compression-before-encryption-leak-info-about-the-input?utm_source=chatgpt.com" class="flex h-4.5 overflow-hidden  [... cut ...]
    Limit attacker interaction (rate‑limit, CAPTCHA) to stop them gathering thousands of samples needed for reliable inference.
    sidechannel.blog

5. Take‑away

Strong encryption does not automatically give strong privacy if earlier stages of the pipeline leak metadata like size. Compression is predictably size‑dependent, so whenever an attacker can measure ciphertext lengths—and especially when they can influence part of the plaintext—you risk exposing the very secrets you tried to hide. The safest rule is simple: never compress attacker‑mixed secrets before you encrypt them, or pad aggressively when you must.

loftafi, 2025-07-22:
The bitchat code (when I reviewed it last week), does not compress the headers, only the messages content. Why do you think bit chat would be vulnerable to this style of attack given that it only compresses the message contents?

hironichu, 2025-07-22:
You could have stopped at `'m not good at explaining the background behind this,` and not published this issue, but no, you decided to waste everybody's time with your LLM explanation.

loftafi, 2025-07-22:
> Because compression ALWAYS leaks entropy. If you send multiple same emoji for example, it will be compressed to one which reveals a lot about the plaintext. ChatGPT explained it too in the text I sent you above...

If multiple emoji's were compressed to one emoji, and then encrypted, how would that (by itself) help an attacker?  I am going to assume at this point I am talking to a bot and not a real person.

hironichu, 2025-07-22:
> If multiple emoji's were compressed to one emoji, and then encrypted, how would that (by itself) help an attacker? I am going to assume at this point I am talking to a bot and not a real person.

Sadly this repo is starting to fill up with spam...

loftafi, 2025-07-22:
I wanted to give the OP the benefit of the doubt. But I think you are right.

hironichu, 2025-07-23:
I understand that perhaps you may know more about this topic. However,

`Bitchat` is an amazing project, getting flooded with feedback and ideas on security, privacy, and QoL features.

This issue (like many others) reflect pure amateurism and childish behavior, if you were a `cryptographic expert` in your own native language then please do express yourself, present some article and a proper reasoning on why `bitchat` must imperatively NOT use compression. Do it in your language with your knowledge and people will figure out how to translate and understand.

> Also, like I added in the second reply: Other messengers ALWAYS use padding, that's one of the most important steps to ensure privacy. You NEVER want to compress plaintext, but instead you always randomise it OR bring it to a fixed size by always padding it to a full block. 🤦🏼‍♂️

If you read the code you'd know that the protocol does include optimized block padding when encoding each packet, that was made to strengthen the network analysis risk, still quite not 100% safe but it's there, since you mentioned it.

```swift
        // Try to compress payload if beneficial
        var payload = packet.payload
        var originalPayloadSize: UInt16? = nil
        var isCompressed = false
        
        if CompressionUtil.shouldCompress(payload) {
            if let compressedPayload = CompressionUtil.compress(payload) {
                // Store original size for decompression (2 bytes after payload)
                originalPayloadSize = UInt16(payload.count)
                payload = compressedPayload
                isCompressed = true
                
            } else {
            }
        } else {
        }
....     
        // Apply padding to standard block sizes for traffic analysis resistance
        let optimalSize = MessagePadding.optimalBlockSize(for: data.count)
        let paddedData = MessagePadding.pad(data, toSize: optimalSize)
        
```

> Because compression ALWAYS leaks entropy. If you send multiple same emoji for example, it will be compressed to one which reveals a lot about the plaintext. ChatGPT explained it too in the text I sent you above...
> 

And the compression still try to compress if the entropy stays high, which lowers the risk of guessing when iterating 

```swift
    // Helper to check if compression is worth it
    static func shouldCompress(_ data: Data) -> Bool {
        // Don't compress if:
        // 1. Data is too small
        // 2. Data appears to be already compressed (high entropy)
        guard data.count >= compressionThreshold else { return false }
        
        // Simple entropy check - count unique bytes
        var byteFrequency = [UInt8: Int]()
        for byte in data {
            byteFrequency[byte, default: 0] += 1
        }
        
        // If we have very high byte diversity, data is likely already compressed
        let uniqueByteRatio = Double(byteFrequency.count) / Double(min(data.count, 256))
        return uniqueByteRatio < 0.9 // Compress if less than 90% unique bytes
    }
```
## 🤯🤯🤯🤯 Bitchat isnt as secure as Signal / Telegram or any other app 🤯🤯🤯🤯

> Also, like I added in the second reply: Other messengers ALWAYS use padding, that's one of the most important steps to ensure privacy. You NEVER want to compress plaintext, but instead you always randomise it OR bring it to a fixed size by always padding it to a full block. 🤦🏼‍♂️

Guess what?  Bluetooth Low Energy was NOT made at all for that purposes, the stack does include encrypted channel between devices (L2CAP) but it requires pairing in order to send more heavy payloads in an encrypted / "reliable" connection.

Bitchat is not like other messaging app, it's not there *yet* but giving PR and code reviews helps, if you are an expert, then go on, gather your knowledge and do a proper security analysis of the code and give it a detailed report, in your native language if you must.

See, i am not an expert, but i love coding, i've made a certain numbered of throwaway project on my Github with shitty code, so i spent a whole day reading up online about the topic of this issue. No chatGPT involved and I learned a lot.

An expert presents an issue with proper research in the code to point out precisely what could/should be improved/changed.

>  I'm correct while you don't understand anything about cryptography.

The classic : i'm an expert and you're not so i'm right and you're wrong.

I may be wrong, i may make mistakes, but please don't use ChatGPT next time you want to explain a security problem, we've seen what this has done to the world of security 
[this is a huge problem for cybersecurity... - LowLevel](https://www.youtube.com/watch?v=xy-u1evNmVo)