permissionlesstech/bitchat #299: Request for private TestFlight invitation, please

chorns7, 2025-07-22:
Hello permissionless-tech team[,]  
I hope you’re doing well[,] My name is MhmdCasta and I’m very interested in testing the iOS beta of Bitchat[,]
Unfortunately the public TestFlight beta is currently full and I can’t join via the public link[,]
Could you please add me to the TestFlight testers group or provide a working redeem code[,] I would greatly appreciate it[,]
Thank you for your time and for this awesome project[,]  
Best regards[,]  
MhmdCasta
🤙🏻

ryannair05, 2025-07-22:
I put my fork of Bitchat in this app if you would like to test, just select "N/A" on the welcome screen: https://apple.co/3F90IYQ

AEROENG1081, 2025-07-23:
I would like to test Bitchat too, but I couldn't. I need an invitation to do so. Can someone help me to get one?

anynon, 2025-07-23:
> Can someone help me to get one?

There's a link above to try it out fully. It's as good or better than testflight