permissionlesstech/bitchat #399: feat req: Peer-to-Peer Video Calls via WebRTC using bitChat Protocol for SDP Exchange

abhiram-ar, 2025-08-04:
### The Problem:
WebRTC is known for enabling peer-to-peer real-time communication, but there’s a catch:
It still relies on a central signaling server to exchange session details (SDP) before the connection is established.
This dependency prevents WebRTC from being truly decentralized.

### The Solution:
- Merge WebRTC with BitChat 
- Use BitChat to exchange WebRTC handshake data (SDPs) over Bluetooth.
-  Zero servers. Zero internet. Pure device-to-device magic.

### Why this matters:
- Enables high-quality P2P video calls without any central servers.
- Works in offline environments (LAN networks in companies, hospitals, malls, disaster zones).
- Enhances privacy since no signaling metadata goes through a server.

Aswinkl1, 2025-08-04:
cool idea

abhiram-ar, 2025-08-17:
@artiomborovinskii I will look into this