permissionlesstech/bitchat #508: Feature Request | Meshtastic Hand off

Luckst4r, 2025-08-24:
### Feature Request: Integrate Meshtastic Handoff for Messages When No Bluetooth Peers Are Available

#### Description
As a user of BitChat, I would like the app to automatically hand off messages to nearby Meshtastic nodes via Bluetooth if no direct Bluetooth peers are detected. This would allow messages to be broadcast over the Meshtastic mesh network (using LoRa for longer-range communication), extending the app's usability in scenarios where direct BT connections are unavailable, such as in remote areas or with sparse user density.

#### Motivation
BitChat already provides decentralized, peer-to-peer messaging over Bluetooth, which is great for short-range, offline communication. However, Bluetooth has limited range (typically 10-100 meters). Meshtastic, an open-source mesh networking project using LoRa radios, can extend this to kilometers in open terrain. Integrating a handoff mechanism would:
- Improve resilience in low-connectivity environments (e.g., hiking, disaster response, or rural areas).
- Leverage existing Meshtastic hardware that users might already own, without requiring major changes to BitChat's core Bluetooth focus.
- Align with the permissionless, decentralized ethos of the project by bridging short-range and long-range offline networks.

This feature could make BitChat a more versatile tool for off-grid communication, potentially attracting users from the Meshtastic community.

#### Proposed Solution
1. **Detection Logic**:
   - When attempting to send a message, first scan for direct Bluetooth peers (as currently implemented).
   - If no peers are found within a configurable timeout (e.g., 10-30 seconds), scan for nearby Meshtastic nodes that support Bluetooth connectivity (many Meshtastic devices, like those based on ESP32, have BT capabilities).

2. **Connection and Handoff**:
   - Connect via Bluetooth to the nearest/detected Meshtastic node.
   - Serialize the BitChat message (including sender info, timestamp, etc.) and forward it to the Meshtastic node.
   - The Meshtastic node would then broadcast the message over its LoRa mesh network.
   - Optionally, include a protocol for receiving messages back from Meshtastic and relaying them to BitChat users via BT.

3. **Technical Considerations**:
   - **Compatibility**: Use the Meshtastic API or protobuf schema for message formatting to ensure seamless integration. Meshtastic's Python library or similar could be referenced for implementation.
   - **User Configuration**: Add settings in the app for enabling Meshtastic integration, selecting preferred nodes, or setting range thresholds.
   - **Security**: Ensure end-to-end encryption is preserved during handoff (e.g., using BitChat's existing encryption before forwarding).
   - **Fallback**: If no Meshtastic nodes are found, fall back to a "message queued" state or notify the user.
   - **Dependencies**: This might require adding Bluetooth Low Energy (BLE) support if not already present, as Meshtastic often uses BLE for configuration/mobile app integration.

4. **Potential Challenges**:
   - Battery impact from additional scanning.
   - Handling message acknowledgments across networks.
   - Ensuring cross-platform compatibility (Android/iOS).

#### Additional Information
- **Related Projects**: Meshtastic documentation is available at [meshtastic.org](https://meshtastic.org). There might be opportunities for collaboration or reusing code from Meshtastic's mobile apps.
- **Testing**: I'd be happy to test this on my setup (e.g., with an ESP32-based Meshtastic node).
- **Labels**: enhancement, integration

If this aligns with the project's roadmap, I'd love to discuss further or contribute if possible!

ahmedkadd, 2025-08-25:
I would absolutely love to see this, as a fan of both Bitchat and Meshtastic

floatinatme, 2025-08-25:
Like to use meshtastic nodes that have WiFi/bluetooth like a repeater to pass a message between Lora and Bluetooth/wifi clients. It would allow it to be used in situations with little infrastructure. Another technology that looks interesting with this concept is the Halow 801.11ah and its long range.

chrisberryman, 2025-09-10:
Interoperability, forwarding, handoff, translation, etc. would certainly benefit both ecosystems.  I would love to see this worked from both sides.

Luckst4r, 2025-09-10:
I have stood up a local meshtastic WAN around me. Happy to test when available 

"target": "https://github.com/permissionlesstech/bitchat/issues/508#issuecomment-3275047819",
"url": "https://github.com/permissionlesstech/bitchat/issues/508#issuecomment-3275047819",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

evansmj, 2025-09-20:
I have been thinking about this, i am about to implement something.  I thought about using meshtastic, however meshtastic is geared towards 1 phone using real bluetooth (not BLE) to connect to the meshtastic node, and then the mestastic node uses lora to relay messages over long distances to other meshtastic nodes.  But then to receive a message, there is 1 device connected to the receiving meshtastic node.  

I dont like this because my use case is being out on deserted island communities with multiple people in each household.  i dont want each person to have to explicitly connect to the meshtastic node to send and receive messages.  I want anybody in the house to just open bitchat and send a message to the whole neighborhood, or to someone who is out fishing.

So I am thinking to just make a custom firmware for Rak 4630 that will be able to listen to BLE advertisements and then relay that to the household lora, then that gets transmitted to a boat or other household's Rak 4630 which then relays to BLE and the phones receive the message

caveman99, 2026-01-08:
@evansmj A cleaner approach would be to use the same mechanism the [Meshtastic ATAK plugin](https://github.com/meshtastic/ATAK-Plugin) is using. It uses an internal API within the android phone to talk to the connected Meshtastic radio via its native app. So there's no fight over who gets to talk to the BLE UUID and it doesn't need a firmware mod.

Luckst4r, 2026-01-12:
I'm now traveling but will test these soon!

Thanks kindly,

Marshall Long

Strategy | http://01.dev