permissionlesstech/bitchat #543: ******Suggestion (GPS Location)******

boraozver, 2025-08-29:
Your program looks really good. As a suggestion, I recommend adding a feature to send GPS location. For example, in cases such as fire, flood, or earthquake, you can help ensure assistance reaches people by sending their GPS location.

boraozver, 2025-08-29:
like this (41.0569308, 28.9531009), someone can send their coordinates to another person via Bluetooth, and then that person can use Google Maps over the internet to reach them and provide help.

SeamusMullan, 2025-09-15:
eg: i send a message you dont like, you note my geolocation (a marker of my identity). when i clear all my data my location in real life doesnt change, you can then use it to link the blank anon account to me, purely using the location.