permissionlesstech/bitchat #549: Botting issues

thebashforkbomber, 2025-08-30:
[image]

These bots are everywhere. As soon as I open a chat that is empty and say anything containing "semicolon", the user count there jumps to two and it sends that one thing. It spams semicolons everywhere too.
Please, add something like a captcha to resolve this. I don't wanna submit to the network of the semicolon.
[image]

thebashforkbomber, 2025-08-30:
Luckily, #bluetooth is untouched. Or not. I think it only happens in US regions where the bots are running.

thebashforkbomber, 2025-08-30:
Yeah...

[image]

thebashforkbomber, 2025-08-30:
[image]

thebashforkbomber, 2025-08-30:
[image]

nomadram, 2025-08-30:
https://github.com/permissionlesstech/bitchat-android/issues/350

I already tried to describe the problem. Now it's even worse.