permissionlesstech/bitchat #562: [Android] Private channels (/j or /join) are not private; messages are broadcast to the entire Bluetooth mesh

mpmerd, 2025-09-05:
**Bug Description**
When creating or joining a private channel (using `/j` or `/join`) on the Android version, the messages from that channel are not kept private. Instead, they are visible to all devices on the Bluetooth mesh network, including those not in the channel and iOS devices.

**Steps to Reproduce**
1.  Use two Android phones (Samsung F16 and Samsung A23) and one iPhone (iPhone 12 Pro).
2.  Connect all three devices to the same Bluetooth mesh network.
3.  On the Samsung F16 (Android), create or join a private channel with a command like `/j testchannel password123`.
4.  From the Samsung A23 (Android), join the same channel with the same password.
5.  Write a message inside the channel from either Android device.
6.  Observe the iPhone 12 Pro, which has **not** joined the channel.

**Expected Behavior**
*   Messages sent within the private channel should **only** be visible to users who have joined that channel with the correct password.
*   The user on the iOS device (or any other device not in the channel) should not be able to read the channel's messages.

**Actual Behavior**
*   Messages from the "private" channel are visible to **all** devices on the mesh network, regardless of whether they have joined the channel or not. The iOS device can read the entire conversation without needing the password.

**Test Environment/Devices:**
*   Android Device 1: Samsung F16
*   Android Device 2: Samsung A23
*   iOS Device: iPhone 12 Pro
*   Network Type: Bluetooth Mesh

**Additional Context**
*   The concept of private channels is fantastic for organizing conversations, but currently, the privacy feature does not seem to be working correctly on Android.
*   I would love to see this feature implemented and working correctly on the iOS app as well, but this current bug on Android suggests the channel system is not properly isolating traffic. This is a critical privacy concern.

**Priority**
High (This is a severe privacy issue that affects a core functionality of the app).

MutantRabbit767, 2025-09-06:
hopefully that's fixed [here](https://github.com/permissionlesstech/bitchat-android/commit/4b81b7f97af5ec2198495bba600fe7aa4b9267cc).

Kalandango, 2025-09-17:
Tried today ( Sep 17th , 2025 ) , and confirm the issue on Android : sending message to created channel ( /j ) delivers it to current Location channel .
Version 1.3.1 ( Updated on Sep 15, 2025 )

naterutland, 2025-09-24:
I tested with 2 android devices locally on BLE with same results. 24 SEP 2025