permissionlesstech/bitchat #601: Idea: Simple Relay Service for Extending Range

theklausklausen, 2025-09-13:
This is more of an idea, and I’d like to hear your thoughts on whether it could be useful, feasible, or worth pursuing.

Similar to LoRaWAN, I was thinking about a small relay service that I could run in my homelab to help extend the effective range. From what I’ve seen, especially in urban areas, the tool isn’t used very frequently yet, so the range can feel quite limited.

The idea would be to set up a lightweight relay (maybe running in Docker) with just a Bluetooth device attached, whose only purpose is to relay messages. I don’t have a spare phone to dedicate to this, so I was wondering if a minimal setup like this could fill the gap.

This is just a quick thought, and I’m not sure if it would actually be helpful, practical, or secure.
What do you think—does this sound like a direction worth exploring?

r1b2ns, 2025-09-14:
“I think it is a very good idea. We can provide a way to run it on Arduinos, but I have a concern about man-in-the-middle attacks. It could also be used for censorship

KevFlo, 2025-09-18:
I was thinking of working on attaching bitchat to a lora mesh service kind of like meshtastic and piggy back on that sort of networking. Although meshtastic already has a text messaging system I don't know if this would be redundant or at least a way to piggy back on meshtastic and extend then range that way.